m5stack/M5AtomS3 #3: multiple definition of TFT Functions

luisllamasbinaburo, 2022-12-26:
### Describe the bug

When I try to compile Arduino Examples, I get this error
"
multiple definition of `chrtbl_f72'; C:\...\Fonts\Font72rle.c:237: first defined here
multiple definition of `chr_f72_20'; C:\...\Fonts\Font72rle.c:28: first defined here
multiple definition of `chr_f72_2D'; C:\...\Fonts\Font72rle.c:32: first defined here
multiple definition of `chr_f72_2E'; C:\...\Fonts\Font72rle.c:37: first defined here
multiple definition of `chr_f72_30'; C:\...\Fonts\Font72rle.c:42: first defined here
multiple definition of `chr_f72_31'; C:\...\Fonts\Font72rle.c:65: first defined here
multiple definition of `chr_f72_32'; C:\...\Fonts\Font72rle.c:81: first defined here
multiple definition of `chr_f72_33'; C:\...\Fonts\Font72rle.c:97: first defined here
multiple definition of `chr_f72_34'; C:\...\Fonts\Font72rle.c:115: first defined here
multiple definition of `chr_f72_35'; C:\...\Fonts\Font72rle.c:133: first defined here
multiple definition of `chr_f72_36'; C:\...\Fonts\Font72rle.c:151: first defined here
multiple definition of `chr_f72_37'; C:\...\Fonts\Font72rle.c:173: first defined here
multiple definition of `chr_f72_38'; C:\...\Fonts\Font72rle.c:187: first defined here
multiple definition of `chr_f72_39'; C:\...\Fonts\Font72rle.c:209: first defined here
multiple definition of `chr_f72_3A'; C:\...\Fonts\Font72rle.c:231: first defined here
multiple definition of `widtbl_f72'; C:\...\Fonts\Font72rle.c:10: first defined here
multiple definition of `TFT_eSPI::readByte()'; C:\...\M5AtomS3-main\utility\In_eSPI.cpp.o:c:\...\src\utility\In_eSPI_ESP32_S3.cpp:96: first defined here
multiple definition of `TFT_eSPI::pushBlock(unsigned short, unsigned int)'; C:\...\M5AtomS3-main\utility\In_eSPI.cpp.o:c:\...\src\utility\In_eSPI_ESP32_S3.cpp:254: first defined here
multiple definition of `TFT_eSPI::pushSwapBytePixels(void const*, unsigned int)'; C:\...\M5AtomS3-main\utility\In_eSPI.cpp.o:c:\...\src\utility\In_eSPI_ESP32_S3.cpp:306: first defined here
multiple definition of `TFT_eSPI::pushPixels(void const*, unsigned int)'; C:\...\M5AtomS3-main\utility\In_eSPI.cpp.o:c:\...\src\utility\In_eSPI_ESP32_S3.cpp:395: first defined here
multiple definition of `dmaHAL'; C:\...\M5AtomS3-main\utility\In_eSPI.cpp.o:c:\...\src\utility\In_eSPI_ESP32_S3.cpp:38: first defined here
multiple definition of `TFT_eSPI::dmaBusy()'; C:\...\M5AtomS3-main\utility\In_eSPI.cpp.o:c:\...\src\utility\In_eSPI_ESP32_S3.cpp:652: first defined here
multiple definition of `TFT_eSPI::dmaWait()'; C:\...\M5AtomS3-main\utility\In_eSPI.cpp.o:c:\...\src\utility\In_eSPI_ESP32_S3.cpp:672: first defined here
multiple definition of `TFT_eSPI::pushPixelsDMA(unsigned short*, unsigned int)'; C:\...\M5AtomS3-main\utility\In_eSPI.cpp.o:c:\...\src\utility\In_eSPI_ESP32_S3.cpp:689: first defined here
multiple definition of `TFT_eSPI::pushImageDMA(int, int, int, int, unsigned short const*)'; C:\...\M5AtomS3-main\utility\In_eSPI.cpp.o:c:\...\src\utility\In_eSPI_ESP32_S3.cpp:721: first defined here
multiple definition of `TFT_eSPI::pushImageDMA(int, int, int, int, unsigned short*, unsigned short*)'; C:\...\M5AtomS3-main\utility\In_eSPI.cpp.o:c:\...\src\utility\In_eSPI_ESP32_S3.cpp:752: first defined here
multiple definition of `dc_callback(spi_transaction_t*)'; C:\...\M5AtomS3-main\utility\In_eSPI.cpp.o:c:\...\src\utility\In_eSPI_ESP32_S3.cpp:841: first defined here
multiple definition of `spi_host'; C:\...\M5AtomS3-main\utility\In_eSPI.cpp.o:c:\...\src\utility\In_eSPI_ESP32_S3.cpp:55: first defined here
multiple definition of `TFT_eSPI::initDMA(bool)'; C:\...\M5AtomS3-main\utility\In_eSPI.cpp.o:c:\...\src\utility\In_eSPI_ESP32_S3.cpp:853: first defined here
multiple definition of `TFT_eSPI::deInitDMA()'; C:\...\M5AtomS3-main\utility\In_eSPI.cpp.o:c:\...\src\utility\In_eSPI_ESP32_S3.cpp:899: first defined here
multiple definition of `spi'; C:\...\M5AtomS3-main\utility\In_eSPI.cpp.o:c:\...\src\utility\In_eSPI_ESP32_S3.cpp:31: first defined here"

### To reproduce

1. Open any Example
2. Verify skecth

### Expected behavior

Skecth should compile correctly.

### Environment

- OS: Windows 10
- IDE &IDE Version: Arduino IDE 1.6.x or Arduino IDE 2.x
- Repository Version: First release (it doesn't have tag)

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Stack/issues?q=)
- [X] My report contains all necessary details

IcingTomato, 2022-12-30:
Would you like to use PlatformIO and one more try?

Tinyu-Zhao, 2023-01-03:
Fixed https://github.com/m5stack/M5AtomS3/commit/3fa7452ef0e8047c4d70cd2444c96cebbfa2538c

closed 2023-01-03 by Tinyu-Zhao