m5stack/M5AtomS3 #8: First attempt with Atom S3 - Arduino - M5AtomS3.h "no file or directory"

MDAR, 2023-01-05:
### Describe the bug

I have been using the M5StickC and M5StickC-PLUS with Arduino for over a year with a lot of success.

I thought I would like to try the Atom S3 as a smaller device for the same MQTT and I2C tasks,

I have selected M5Stack-AtomS3 from the M5Stack Arduino board manager (after adding the URL for the library)

Pasting your example Button code into Arduino and trying to flash to Com3 

All I get is an error message saying -
```
Arduino: 1.8.13 (Windows 10), Board: "M5Stack-ATOM, Default, 1500000, None"

RepeatTimer:15:22: fatal error: M5AtomS3.h: No such file or directory

compilation terminated.

exit status 1

M5AtomS3.h: No such file or directory

```

### To reproduce

1. On a Windows 10 with Arduino 1.8.13
2. Using the sample Button.ino code
```
/*
*******************************************************************************
* Copyright (c) 2022 by M5Stack
*                  Equipped with M5AtomS3 sample source code
*                          配套  M5AtomS3 示例源代码
* Visit for more information: https://docs.m5stack.com/en/core/AtomS3
* 获取更多资料请访问: https://docs.m5stack.com/zh_CN/core/AtomS3
*
* Describe: Button example.  按键示例
* Date: 2022/12/18
*******************************************************************************
Press button to display the corresponding output on the screen and
USBserial. 按下按键，在屏幕和串口上显示相应输出
*/
#include <M5AtomS3.h>
/* After M5AtomS3 is started or reset
  the program in the setUp () function will be run, and this part will only be
  run once. 在 M5AtomS3
  启动或者复位后，即会开始执行setup()函数中的程序，该部分只会执行一次。 */
void setup() {
    M5.begin(true, true, false, false);  // Init M5AtomS3.  初始化 M5AtomS3
    M5.Lcd.println("Pls Press Btn");
}

/* After the program in setup() runs, it runs the program in loop()
The loop() function is an infinite loop in which the program runs repeatedly
在setup()函数中的程序执行完后，会接着执行loop()函数中的程序
loop()函数是一个死循环，其中的程序会不断的重复运行 */
void loop() {
    M5.update();  // Read the press state of the key.  读取按键 A, B, C 的状态
    if (M5.Btn.wasReleased() || M5.Btn.pressedFor(1000)) {
        USBSerial.print('A');
        M5.Lcd.print("A");
    }
}
```

### Expected behavior

I was kind of hoping it would just work

### Environment

- OS: Windows 10
- IDE & IDE Version: Arduino 1.8.13
- Repository Version: https://github.com/m5stack/M5AtomS3/blob/main/examples/Basics/Button/Button.ino

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Stack/issues?q=)
- [X] My report contains all necessary details

Sarah-C, 2023-01-06:
What you downloaded was the RISC chip compiler for the ESP32 chip, and M5Stacks's core libraries - like the "USBSerial" object the examples use.

You're missing the M5AtomS3 software library! (It's the built that codes for the TFT LCD screen, inertial unit, buttons, pin names, and alllllll that).. which is exactly what this GitHub contains. YAY!

On the main Github page here, https://github.com/m5stack/M5AtomS3/  , near the top... Click the blue button that says "<> Code",  in the menu that appears click "Download ZIP".

In your browsers download folder, unzip the "M5AtomsS3-main.zip" file to you Libraries file - it's in your documents folder...... /Documents/Arduino/Libraries/  ..... there'll be a bunch of folders there already probably!

Close and reopen Arduino IDE.

The examples will now work, because the "Include AtomS3.h" will now work - that and the CPP and other files are what's in this library.

ALSO - putting the Library in this folder, you get all the examples BUILT IN! Yay!
From "File..... Examples....." scroll down........ and down..... until you reach "M5AtomS3" and all the examples in this github are there, for you to open and run!

Tinyu-Zhao, 2023-01-09:
I have now submitted the AtomS3 library to the Arduino library and it will be searchable in a few hours.

MDAR, 2023-01-09:
@Sarah-C 
@Tinyu-Zhao 

Thank you both for your help.

I'm really looking forward to seeing what I can do with this little device.

3110, 2023-01-12:
> I have now submitted the AtomS3 library to the Arduino library and it will be searchable in a few hours.

@Tinyu-Zhao AtomS3 library is not yet searchable.

Tinyu-Zhao, 2023-01-14:
I'm working on [it](https://github.com/arduino/library-registry/pull/2352#issuecomment-1379048774).

Tinyu-Zhao, 2023-01-15:
> > I have now submitted the AtomS3 library to the Arduino library and it will be searchable in a few hours.
> 
> @Tinyu-Zhao AtomS3 library is not yet searchable.

Fixed.

closed 2023-01-15 by Tinyu-Zhao

MDAR, 2023-01-15:
Thank you @Tinyu-Zhao 
I'm looking forward to seeing what I can do with it.