m5stack/M5AtomS3 #10: AtomS3 fails to upload programs

MrStraws, 2023-01-07:
After cleanly compiling any of the AtomS3examples they fail to upload giving the following error:
A fatal error occurred: Invalid head of packet (0x47)

Tinyu-Zhao, 2023-01-09:
Please provide us with more information so that we can identify the problem

MrStraws, 2023-01-09:
Thank you for the prompt reply. The attached file is a copy of error
messages generated upon the attempted upload to the Atom S3 from the
Arduino 1.8.12 IDE. I compiled the 'GPIOInterrupt.ino" file from the Atom
S3 examples on Github.The compilation apparently completed OK as shown in
the attached file. One of the error messages deals with a missing crystal
on the new AtomS3 in question. This AtomS3 appears to power up OK and runs
through the initial 'demoes' of WiFi, I2C Scanning, UART Mon, PORTA PWM,
etc. I held the reset button on the side of the Atom until a green internal
LED turned on and then compiled and tried to upload again with no success.
I've had no upload success with files I've tried.
Any further help you can offer is appreciated.
Bob Swarts

-- 
Bob

Arduino: 1.8.12 (Windows 10), Board: "M5Stack-ATOM, Default, 115200, None"

Linking everything together...
"C:\\Users\\rswar\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-gcc" -nostdlib "-LC:\\Users\\rswar\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/lib" "-LC:\\Users\\rswar\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/ld" -T esp32_out.ld -T esp32.project.ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.libgcc.ld -T esp32.rom.spiram_incompatible_fns.ld -u esp_app_desc -u ld_include_panic_highint_hdl -u call_user_start_cpu0 -Wl,--gc-sections -Wl,-static -Wl,--undefined=uxTopUsedPriority -u __cxa_guard_dummy -u __cxx_fatal_exception -Wl,--start-group "C:\\Users\\rswar\\AppData\\Local\\Temp\\arduino_build_408282\\sketch\\GPIOInterrupt.ino.cpp.o" "C:\\Users\\rswar\\AppData\\Local\\Temp\\arduino_build_408282\\core\\core.a" -lgcc -lesp_websocket_client -lwpa2 -ldetection -lesp_https_server -lwps -lhal -lconsole -lpe -lsoc -lsdmmc -lpthread -llog -lesp_http_client -ljson -lmesh -lesp32-camera -lnet80211 -lwpa_supplicant -lc -lmqtt -lcxx -lesp_https_ota -lulp -lefuse -lpp -lmdns -lbt -lwpa -lspiffs -lheap -limage_util -lunity -lrtc -lmbedtls -lface_recognition -lnghttp -ljsmn -lopenssl -lcore -lfatfs -lm -lprotocomm -lsmartconfig -lxtensa-debug-module -ldl -lesp_event -lesp-tls -lfd -lespcoredump -lesp_http_server -lfr -lsmartconfig_ack -lwear_levelling -ltcp_transport -llwip -lphy -lvfs -lcoap -lesp32 -llibsodium -lbootloader_support -ldriver -lcoexist -lasio -lod -lmicro-ecc -lesp_ringbuf -ldetection_cat_face -lapp_update -lespnow -lface_detection -lapp_trace -lnewlib -lbtdm_app -lwifi_provisioning -lfreertos -lfreemodbus -lethernet -lnvs_flash -lspi_flash -lc_nano -lexpat -lfb_gfx -lprotobuf-c -lesp_adc_cal -ltcpip_adapter -lstdc++ -Wl,--end-group -Wl,-EL -o "C:\\Users\\rswar\\AppData\\Local\\Temp\\arduino_build_408282/GPIOInterrupt.ino.elf"
"C:\\Users\\rswar\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esptool_py\\3.0.0/esptool.exe" --chip esp32 elf2image --flash_mode dio --flash_freq 80m --flash_size 4MB -o "C:\\Users\\rswar\\AppData\\Local\\Temp\\arduino_build_408282/GPIOInterrupt.ino.bin" "C:\\Users\\rswar\\AppData\\Local\\Temp\\arduino_build_408282/GPIOInterrupt.ino.elf"
esptool.py v3.0-dev

"C:\\Users\\rswar\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/gen_esp32part.exe" -q "C:\\Users\\rswar\\AppData\\Local\\Temp\\arduino_build_408282/partitions.csv" "C:\\Users\\rswar\\AppData\\Local\\Temp\\arduino_build_408282/GPIOInterrupt.ino.partitions.bin"
"C:\\Users\\rswar\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-size" -A "C:\\Users\\rswar\\AppData\\Local\\Temp\\arduino_build_408282/GPIOInterrupt.ino.elf"
Sketch uses 205366 bytes (15%) of program storage space. Maximum is 1310720 bytes.
Global variables use 13920 bytes (4%) of dynamic memory, leaving 313760 bytes for local variables. Maximum is 327680 bytes.
esptool.py v3.0-dev
Serial port COM4
Connecting...
WARNING: This chip doesn't appear to be a ESP32 (date codes 0x03700000:0x00000000). Probably it is unsupported by this version of esptool.
Chip is ESP32-D0WDQ6 (revision 0)
Features: WiFi, BT, Dual Core, Coding Scheme None
WARNING: Detected crystal freq 0.00MHz is quite different to normalized freq 26MHz. Unsupported crystal in use?
Crystal is 26MHz
MAC: 00:00:00:00:00:00
Uploading stub...
Running stub...

A fatal error occurred: Invalid head of packet (0x47)
A fatal error occurred: Invalid head of packet (0x47)

Tinyu-Zhao, 2023-01-11:
Can you give the error message and the example GPIOInterrupt.ino that you used?

MrStraws, 2023-01-12:
I have attached both files asked for. The error file contains the Arduino
compilation steps and at the end of the file are the error messages the
compiler outputs. There are also several references to crystals and crystal
frequencies. They obviously refer to some hardware problems that lead me to
believe that I won't be able to upload any Arduino files to the Atom S3 .
Any help you can offer is appreciated.

Arduino: 1.8.12 (Windows 10), Board: "M5Stack-ATOM, Default, 115200, None"

Linking everything together...
"C:\\Users\\rswar\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-gcc" -nostdlib "-LC:\\Users\\rswar\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/lib" "-LC:\\Users\\rswar\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/ld" -T esp32_out.ld -T esp32.project.ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.libgcc.ld -T esp32.rom.spiram_incompatible_fns.ld -u esp_app_desc -u ld_include_panic_highint_hdl -u call_user_start_cpu0 -Wl,--gc-sections -Wl,-static -Wl,--undefined=uxTopUsedPriority -u __cxa_guard_dummy -u __cxx_fatal_exception -Wl,--start-group "C:\\Users\\rswar\\AppData\\Local\\Temp\\arduino_build_408282\\sketch\\GPIOInterrupt.ino.cpp.o" "C:\\Users\\rswar\\AppData\\Local\\Temp\\arduino_build_408282\\core\\core.a" -lgcc -lesp_websocket_client -lwpa2 -ldetection -lesp_https_server -lwps -lhal -lconsole -lpe -lsoc -lsdmmc -lpthread -llog -lesp_http_client -ljson -lmesh -lesp32-camera -lnet80211 -lwpa_supplicant -lc -lmqtt -lcxx -lesp_https_ota -lulp -lefuse -lpp -lmdns -lbt -lwpa -lspiffs -lheap -limage_util -lunity -lrtc -lmbedtls -lface_recognition -lnghttp -ljsmn -lopenssl -lcore -lfatfs -lm -lprotocomm -lsmartconfig -lxtensa-debug-module -ldl -lesp_event -lesp-tls -lfd -lespcoredump -lesp_http_server -lfr -lsmartconfig_ack -lwear_levelling -ltcp_transport -llwip -lphy -lvfs -lcoap -lesp32 -llibsodium -lbootloader_support -ldriver -lcoexist -lasio -lod -lmicro-ecc -lesp_ringbuf -ldetection_cat_face -lapp_update -lespnow -lface_detection -lapp_trace -lnewlib -lbtdm_app -lwifi_provisioning -lfreertos -lfreemodbus -lethernet -lnvs_flash -lspi_flash -lc_nano -lexpat -lfb_gfx -lprotobuf-c -lesp_adc_cal -ltcpip_adapter -lstdc++ -Wl,--end-group -Wl,-EL -o "C:\\Users\\rswar\\AppData\\Local\\Temp\\arduino_build_408282/GPIOInterrupt.ino.elf"
"C:\\Users\\rswar\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esptool_py\\3.0.0/esptool.exe" --chip esp32 elf2image --flash_mode dio --flash_freq 80m --flash_size 4MB -o "C:\\Users\\rswar\\AppData\\Local\\Temp\\arduino_build_408282/GPIOInterrupt.ino.bin" "C:\\Users\\rswar\\AppData\\Local\\Temp\\arduino_build_408282/GPIOInterrupt.ino.elf"
esptool.py v3.0-dev

"C:\\Users\\rswar\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/gen_esp32part.exe" -q "C:\\Users\\rswar\\AppData\\Local\\Temp\\arduino_build_408282/partitions.csv" "C:\\Users\\rswar\\AppData\\Local\\Temp\\arduino_build_408282/GPIOInterrupt.ino.partitions.bin"
"C:\\Users\\rswar\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-size" -A "C:\\Users\\rswar\\AppData\\Local\\Temp\\arduino_build_408282/GPIOInterrupt.ino.elf"
Sketch uses 205366 bytes (15%) of program storage space. Maximum is 1310720 bytes.
Global variables use 13920 bytes (4%) of dynamic memory, leaving 313760 bytes for local variables. Maximum is 327680 bytes.
esptool.py v3.0-dev
Serial port COM4
Connecting...
WARNING: This chip doesn't appear to be a ESP32 (date codes 0x03700000:0x00000000). Probably it is unsupported by this version of esptool.
Chip is ESP32-D0WDQ6 (revision 0)
Features: WiFi, BT, Dual Core, Coding Scheme None
WARNING: Detected crystal freq 0.00MHz is quite different to normalized freq 26MHz. Unsupported crystal in use?
Crystal is 26MHz
MAC: 00:00:00:00:00:00
Uploading stub...
Running stub...

A fatal error occurred: Invalid head of packet (0x47)
A fatal error occurred: Invalid head of packet (0x47)

Tinyu-Zhao, 2023-01-14:
Please update the Boards Manager to 2.0.6 and use M5AtomS3 Development to compile any AtomS3 projects.

Tutorials:https://docs.m5stack.com/en/quick_start/arduino
[image]

closed 2023-01-15 by Tinyu-Zhao