m5stack/M5AtomS3 #12: Try to start w/ M5AtomS3

ckuehnel, 2023-01-12:
### Describe the bug

I got my M5StackS3 today and tried to start w/ a simple program:

#include <M5AtomS3.h> // https://github.com/m5stack/M5AtomS3

#define DEVICE "M5AtomS3"

void setup() 
{
  M5.begin(true, true, false, false); // LCD, Serial, I2C, LED
  M5.Lcd.println(DEVICE);
  Serial.println(DEVICE);
}

void loop() {}

### To reproduce

I got compiler errors due to serial communication. I found _Serial_ in M5AtomS3.h and _USBSerial_ in M5AtomS3.cpp. After changing it to _Serial_ I got display errors. Because the AtomS3 has no WS2812 I commented these lines in utility/LED_DisPlay.*
After that, I got no compiler errors in the legacy IDE. In IDE 2.0 I get the message: 
  exit status 1
  Compilation error: exit status 1
If I try to upload the compiled code in the legacy IDE, there is a device mismatch ESP32-S3 detected but ESP32 in the code used.
I do not understand what is wrong here.

### Expected behavior

I expect an output of the String M5AtomS3 on LCD and via serial communication @ 115200 bps. My configuration is below.

### Screenshots

[image]

### Environment

- OS:
- IDE &IDE Version:
- Repository Version:

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Stack/issues?q=)
- [X] My report contains all necessary details

Tinyu-Zhao, 2023-01-14:
[image]
Please update the Development Edition Manager to 2.0.6 and use M5AtomS3 Development Edition to compile any AtomS3 projects.

ckuehnel, 2023-01-14:
It works after updating.Thanks

closed 2023-01-14 by Tinyu-Zhao