m5stack/M5AtomS3 #13: I2C on Grove

ckuehnel, 2023-01-14:
### Describe the bug

I found w/ I2C Scanner @ SCL =38 and SDA=39 an I2c device w/ Addr 68.
I want to connect Grove sensor to the grove port and tried SCL=1 SDA = 2 but cannot find a sensor.  
Because I have no schematic for the ATOMS3 I need your help.

### To reproduce

/*
*******************************************************************************
* Copyright (c) 2021 by M5Stack
*                  Equipped with M5AtomS3 sample source code
*                          配套  M5AtomS3 示例源代码
* Visit for more information: https://docs.m5stack.com/en/core/AtomS3
* 获取更多资料请访问: https://docs.m5stack.com/zh_CN/core/AtomS3
*
* Describe: I2C Scanner.  I2C探测
* Date: 2021/7/27
*******************************************************************************
  This program scans the addresses 1-127 continuosly and shows the devices found
on the TFT. 该程序连续扫描地址 1-127 并显示在外部(内部)I2C发现的设备。
*/
#include <M5AtomS3.h>

void setup() {
    M5.begin(true, true, false, // I2C Enable set to false for separate initialization on next line
             false);  // Init M5AtomS3(Initialization of external I2C is also
                      // included).  M5AtomS3(初始化外部I2C也包含在内)
    Wire.begin(38, 39); //Detect internal I2C, if this sentence is not added,
    // it will detect external I2C.  检测内部I2C,若不加此句为检测外部I2C
    M5.Lcd.println("M5AtomS3 I2C Tester");  // Print a string on the screen.
                                            // 在屏幕上打印字符串
    delay(3000);
}

### Expected behavior

I2C interface @ Port.A

### Environment

- OS:
- IDE &IDE Version:
- Repository Version:

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Stack/issues?q=)
- [X] My report contains all necessary details

Tinyu-Zhao, 2023-01-15:
You can try Wire.begin(1, 2);

ckuehnel, 2023-01-15:
Solved. Wire1(2,1) works.Dr. Claus Kühnel - Talstr. 13b - CH-8852 AltendorfMobile +41-76-3960801

closed 2023-01-15 by Tinyu-Zhao