m5stack/M5AtomS3 #15: In the AtomS3 examples what does the LED.ino sketch do?

teastainGit, 2023-02-13:
### Describe the bug

I run the LED.ino sketch and it does nothing.
I think maybe sketch from old Aton Matrix?
If so, please remove as it confuses new users.

### To reproduce

upload sketch

### Expected behavior

Do something

### Environment

- OS:
- IDE &IDE Version:
- Repository Version:

### Additional context

Your over-bearing issue format is annoying if I have a simple question!

It is very thorough.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Stack/issues?q=)
- [X] My report contains all necessary details

teastainGit, 2023-02-27:
No response so please close.

closed 2023-02-27 by teastainGit

Tinyu-Zhao, 2023-03-03:
Sorry, my negligence is not clearly marked, only AtomS3Lite has RGB lights.