m5stack/M5AtomS3 #16: RGB colours are swapped when using M5.Lcd.setTextColor(RED);

teastainGit, 2023-02-13:
When using M5.Lcd.setTextColor(RED);
GREEN is of course green, but RED text shows blue and BLUE text shows as red on the LCD.
There are no examples in the Examples repository (that I could find) that show LCD text.
What method of displaying text in different fonts are available for AtomS3?
[image]

Also:
How can I stop these pragma messages? I try to run a clean setup and orange messages are alarming, but these are harmless.

In file included from /Users/terrystainton/Documents/Arduino/libraries/FastLED/src/FastLED.h:67,
                 from /Users/terrystainton/Documents/Arduino/libraries/M5AtomS3/src/utility/LED_DisPlay.h:4,
                 from /Users/terrystainton/Documents/Arduino/libraries/M5AtomS3/src/utility/LED_DisPlay.cpp:1:
/Users/terrystainton/Documents/Arduino/libraries/FastLED/src/fastspi.h:145:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/terrystainton/Documents/Arduino/libraries/FastLED/src/FastLED.h:67,
                 from /Users/terrystainton/Documents/Arduino/libraries/M5AtomS3/src/M5AtomS3.h:8,
                 from /Users/terrystainton/Documents/Arduino/libraries/M5AtomS3/src/M5AtomS3.cpp:5:
/Users/terrystainton/Documents/Arduino/libraries/FastLED/src/fastspi.h:145:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Thanks- Terry !!!

teastainGit, 2023-02-27:
Please answer my Issue.

Also please tell me how to use different fonts, in LCD.Print, such as FreeFonts, etc.

colintd, 2023-04-14:
@teastainGit I hit the same issue.  After a bit of digging I've fixed the color ordering problem in this patch https://github.com/m5stack/M5AtomS3/pull/19

You can apply the same changes to your local copy of the file, and then force a rebuild of the the AtomS3 library.

colintd, 2023-04-14:
I've also just added a fix for a +2,+1 offset in the display position, and totally broken "inverted" rotations.  They are in the same pull request.

Philhoff26, 2023-04-17:
> @colintd  I am new to github, and the AtomS3, which I have been having great fun with, not to arduino.  Where do I look for guidance for WHICH repository I need to download and how I make it active.  I did pull the M5Stack Atom3 repository and added the zip to the IDE, overwrote existing, but the results did not change (still wrong colors and missing pixels)  Any help would be appreciated

colintd, 2023-04-17:
It varies slightly on Windows vs Linux, but on Windows you want to update the two changed source/header files in %SKETCHPATH%\libraries\M5AtomS3 and then delete the objects in %USERDATA%\AppData\Local\Temp\arduino\sketches\{random_string_of_hex}\libraries\M5AtomS3\*.o.  This latter step is important, as otherwise the IDE doesn't always rebuild the library.

colintd, 2023-04-17:
The two files you need can be found by looking at the diff for the pull https://github.com/m5stack/M5AtomS3/pull/19/files.

This shows I've updated: src/utility/In_eSPI_Drivers/ST7789_Rotation.h & src/utility/In_eSPI_Setup.h.

My patch pull request has not yet been accepted, so suggest you:

-  clone my patch repro from here https://github.com/colintd/M5AtomS3/tree/patch-1 
- Find those two changed files
- Copy over the top on the current versions in %SKETCHPATH%\libraries\M5AtomS3

Alternatively, you can just find the two files in your %SKETCHPATH%\libraries\M5AtomS3 directory, then make the small number of edits shown in the diff.

Does that make sense?

Philhoff26, 2023-04-17:
Yes, thank you. That makes good sense. I'll look at the diff files and see what the changes were. I have no problem changing the files, just have to locate the two of them and follow your earlier instructions to get them into my build. I appreciate your time and effort.

I'm a PC specialist with my own business to assist people and small businesses with PC problems.  A programmer since the 70s, retired and enjoy playing with the hardware and software for fun.

Thank you for your patience, I'll let you know if I have any luck in the next day or so.
Phil

⁣Get BlueMail for Android ​

Philhoff26, 2023-04-18:
@ColintdIt worked!!I had to hunt a little bit for the temp files
I found the two headers in:C:\Users\philh\Documents\Arduino\libraries\M5AtomS3\src\utility\In_eSPI_Drivers
C:\Users\philh\Documents\Arduino\libraries\M5AtomS3\src\utility
Overkill, but I edited them in Visual Studio b/c that was the default last time I touched a .h file.
I found the temp files in:C:\Users\philh\AppData\Local\Temp\arduino\sketches\1D7B7328475A2566937BD1C6852BBF11\libraries\M5AtomS3 (My test sketch) and deleted the .o files.

Recompiled my test sketch in IDE 2.0.4, and all predefined colors displayed correctly and the display was left-justified correctly with no missing or cutoff pixels.  Great catch/correction of the error.Thank you again for all your ***@***.***
------------------------------------------
Phil Hoffstadter 
***@***.***

colintd, 2023-04-18:
Glad to hear it all worked, and happy to have helped.  

I think the AtomS3 is a really nice little unit, and getting fixes for these problems makes all the difference.  In particular, the SPI fix means it is a great ethernet powered sensor/display unit when on the AtomPOE base

Tinyu-Zhao, 2023-04-19:
Now I have released a new version, I guess you can see the update on Arduino soon

closed 2023-04-19 by Tinyu-Zhao