m5stack/M5AtomS3 #17: Can we get the source code of the "fancy" factory program?

NodeMCU-Kang, 2023-03-07:
Hi, I just received my AtomS3 ordered a week ago. When I powered on it, the factory program was so impressive. It showed a animated rainbow image, changed to different features by pressing on the LCD button, and the IMU feature demo is very intuitive.

I thought "examples/Basics/FactoryTest" was the source code of the "fancy" factory demo, but it is not. There is rainbow.c, but it is not used by main.cpp. 

Is it possible that M5Stack provide the source code of  the "fancy" factory problem?

hectorri, 2023-03-22:
Please, share code of factory example.

imliubo, 2023-03-23:
Sorry, we will add and update this ASAP.

Tinyu-Zhao, 2023-03-24:
@NodeMCU-Kang @hectorri 
The factory program has been updated.
https://github.com/m5stack/M5AtomS3/commit/57809f946e2ec1543d0c1fc973119eebac18e96c

closed 2023-03-24 by Tinyu-Zhao