m5stack/M5AtomS3 #23: Compilation error when including M5AtomS3.h

scholt, 2023-11-22:
### Describe the bug

I'm trying to get started with the M5 AtomS3 Lite board and to write a simple LED blink program ... unfortunately there are no PINs defined so I tried to use the official sample (https://github.com/m5stack/M5AtomS3/blob/main/examples/Basics/Led/Led.ino) which is linked from the product page of the M5 AtomS3 Lite (https://shop.m5stack.com/products/atoms3-lite-esp32s3-dev-kit).

Unfortunately every time I try to follow the linked sample or even just include the M5AtomS3.h in a sample file which was working before I get the following error:

```
In file included from /Users/username/Documents/Arduino/libraries/FastLED/src/FastLED.h:75,
                 from /Users/username/Documents/Arduino/libraries/M5AtomS3/src/M5AtomS3.h:8,
                 from /Users/username/Documents/Arduino/led_on_off_serial/led_on_off_serial.ino:1:
/Users/username/Documents/Arduino/libraries/FastLED/src/fastspi.h:157:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/username/Documents/Arduino/libraries/FastLED/src/FastLED.h:75,
                 from /Users/username/Documents/Arduino/libraries/M5AtomS3/src/M5AtomS3.h:8,
                 from /Users/username/Documents/Arduino/libraries/M5AtomS3/src/M5AtomS3.cpp:5:
/Users/username/Documents/Arduino/libraries/FastLED/src/fastspi.h:157:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/username/Documents/Arduino/libraries/M5AtomS3/src/M5AtomS3.cpp: In member function 'void M5AtomS3::begin(bool, bool, bool, bool)':
/Users/username/Documents/Arduino/libraries/M5AtomS3/src/M5AtomS3.cpp:20:9: error: 'USBSerial' was not declared in this scope
         USBSerial.begin(115200);
         ^~~~~~~~~
/Users/username/Documents/Arduino/libraries/M5AtomS3/src/M5AtomS3.cpp:20:9: note: suggested alternative: 'Serial'
         USBSerial.begin(115200);
         ^~~~~~~~~
         Serial

exit status 1

Compilation error: exit status 1
```

### To reproduce

1. Open Arduino IDE
2. #include <M5AtomS3.h>
3. Verify Sketch

### Expected behavior

Work like described in the documentation/sample

### Environment

- OS: MacOS 14.0
- IDE &IDE Version: Arduino IDE 2.2.1
- Repository Version: 0.0.3

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Stack/issues?q=)
- [X] My report contains all necessary details

Feryx10, 2023-11-23:
Change the USBSerial object to Serial in M5AtomS3.cpp , and compile.
Example:
```c++
if (USBSerialEnable) {
        Serial.begin(115200);
        Serial.flush();
        delay(1200);
        Serial.println("M5AtomS3 initializing...OK");
    }
``` 
Works

scholt, 2023-11-25:
@Feryx10 thank you very much this works! I hope this gets fixed in the library as the example fresh out of the repository does not work without this fix :)

hallinen, 2023-11-26:
I was having the same problem. I opened M5AtomS3.cpp with a text editor, changed it like you said, but I don't know how to compile the .cpp file.  I saved the .cpp file after I edited it.  I then tried to compile the example "button" from the ATOMS3 basic example and instead of getting the "USBSerial not described in this scope I got a new error...

In file included from /Users/dianehallinen/Documents/Arduino/libraries/FastLED/src/FastLED.h:75,
                 from /Users/dianehallinen/Documents/Arduino/libraries/M5AtomS3/src/M5AtomS3.h:8,
                 from /private/var/folders/tc/p1d_fw296cb18b8wnf6rdm280000gn/T/.arduinoIDE-unsaved20231026-41153-fswrn6.re979/Button/Button.ino:15:
/Users/dianehallinen/Documents/Arduino/libraries/FastLED/src/fastspi.h:157:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/private/var/folders/tc/p1d_fw296cb18b8wnf6rdm280000gn/T/.arduinoIDE-unsaved20231026-41153-fswrn6.re979/Button/Button2.ino:20:6: error: redefinition of 'void setup()'
 void setup() {
      ^~~~~
/private/var/folders/tc/p1d_fw296cb18b8wnf6rdm280000gn/T/.arduinoIDE-unsaved20231026-41153-fswrn6.re979/Button/Button.ino:20:6: note: 'void setup()' previously defined here
 void setup() {
      ^~~~~
/private/var/folders/tc/p1d_fw296cb18b8wnf6rdm280000gn/T/.arduinoIDE-unsaved20231026-41153-fswrn6.re979/Button/Button2.ino:29:6: error: redefinition of 'void loop()'
 void loop() {
      ^~~~
/private/var/folders/tc/p1d_fw296cb18b8wnf6rdm280000gn/T/.arduinoIDE-unsaved20231026-41153-fswrn6.re979/Button/Button.ino:29:6: note: 'void loop()' previously defined here
 void loop() {
      ^~~~

exit status 1

Compilation error: redefinition of 'void setup()'

I'm a very basic arduino person, I've never messed around with the .cpp files.  Should I wait and hope the library gets fixed or am is there something I should try to fix. Thanks so much for posting your reply and asking this question.

I8Pizza2Day, 2023-12-13:
Getting Constant errors trying to setup an AtomS3 lite with both the ENVIII and ENVIIV sensors. I can never seem to get it to Arduino

Tinyu-Zhao, 2024-06-07:
fixed.

closed 2024-06-07 by Tinyu-Zhao