m5stack/M5AtomS3 #27: Error during verify/compile after #include "M5AtomS3.h"

gkowalewski, 2024-08-02:
### Describe the bug

* Installed the newest Arduino IDE 2.3.2 (Mac OS ARM)
* Installed M5AtomS3 library with all dependencies
* M5AtomS3 board selected

In simple code I have following error

#include "M5AtomS3.h"

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}

In file included from /Users/user/projects/Arduino/libraries/FastLED/src/FastLED.h:79,
                 from /Users/user/projects/Arduino/libraries/M5AtomS3/src/./utility/LedDisplay.h:4,
                 from /Users/user/projects/Arduino/libraries/M5AtomS3/src/M5AtomS3.h:6,
                 from /Users/user/projects/Arduino/sketch_aug2a/sketch_aug2a.ino:1:
/Users/user/projects/Arduino/libraries/FastLED/src/fastspi.h:157:23: note: '#pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output'
  157 | #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/user/projects/Arduino/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp: In member function 'void ESP32RMTController::tx_start()':
/Users/user/projects/Arduino/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:326:32: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'mem_rd_rst_n'
  326 |     RMT.chnconf0[mRMT_channel].mem_rd_rst_n = 1;
      |                                ^~~~~~~~~~~~
/Users/user/projects/Arduino/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:327:32: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'mem_rd_rst_n'
  327 |     RMT.chnconf0[mRMT_channel].mem_rd_rst_n = 0;
      |                                ^~~~~~~~~~~~
/Users/user/projects/Arduino/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:328:32: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'apb_mem_rst_n'
  328 |     RMT.chnconf0[mRMT_channel].apb_mem_rst_n = 1;
      |                                ^~~~~~~~~~~~~
/Users/user/projects/Arduino/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:329:32: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'apb_mem_rst_n'
  329 |     RMT.chnconf0[mRMT_channel].apb_mem_rst_n = 0;
      |                                ^~~~~~~~~~~~~
/Users/user/projects/Arduino/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:335:32: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'conf_update_n'
  335 |     RMT.chnconf0[mRMT_channel].conf_update_n = 1;
      |                                ^~~~~~~~~~~~~
/Users/user/projects/Arduino/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:336:32: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'tx_start_n'
  336 |     RMT.chnconf0[mRMT_channel].tx_start_n = 1;
      |                                ^~~~~~~~~~
/Users/user/projects/Arduino/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp: In static member function 'static void ESP32RMTController::doneOnChannel(rmt_channel_t, void*)':
/Users/user/projects/Arduino/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:389:27: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'tx_stop_n'
  389 |     RMT.chnconf0[channel].tx_stop_n = 1;
      |                           ^~~~~~~~~
/Users/user/projects/Arduino/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:390:27: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'conf_update_n'
  390 |     RMT.chnconf0[channel].conf_update_n = 1;
      |                           ^~~~~~~~~~~~~
/Users/user/projects/Arduino/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:392:27: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'mem_rd_rst_n'
  392 |     RMT.chnconf0[channel].mem_rd_rst_n = 1;
      |                           ^~~~~~~~~~~~
/Users/user/projects/Arduino/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:393:27: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'mem_rd_rst_n'
  393 |     RMT.chnconf0[channel].mem_rd_rst_n = 0;
      |                           ^~~~~~~~~~~~
/Users/user/projects/Arduino/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:394:27: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'apb_mem_rst_n'
  394 |     RMT.chnconf0[channel].apb_mem_rst_n = 1;
      |                           ^~~~~~~~~~~~~
/Users/user/projects/Arduino/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:395:27: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'apb_mem_rst_n'
  395 |     RMT.chnconf0[channel].apb_mem_rst_n = 0;
      |                           ^~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1

### To reproduce

as above

### Expected behavior

after #include "M5AtomS3.h" there are no errors while compiling 

### Screenshots

[image]

### Environment

- OS: Sanoma 14.1 Apple M3 
- IDE &IDE Version: 2.3.2
- Repository Version: M3AtomS3 1.0.0

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Stack/issues?q=)
- [X] My report contains all necessary details

gkowalewski, 2024-08-02:
When I also installed M5Stack library (additionally to M5AtomS3) many errors disappeared but it's still active:

In file included from /Users/user/projects/Arduino/libraries/FastLED/src/FastLED.h:79,
                 from /Users/user/projects/Arduino/libraries/M5AtomS3/src/./utility/LedDisplay.h:4,
                 from /Users/user/projects/Arduino/libraries/M5AtomS3/src/M5AtomS3.h:6,
                 from /Users/user/projects/Arduino/sketch_aug2a/sketch_aug2a.ino:1:
/Users/user/projects/Arduino/libraries/FastLED/src/fastspi.h:157:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Sketch uses 392289 bytes (29%) of program storage space. Maximum is 1310720 bytes.
Global variables use 20960 bytes (6%) of dynamic memory, leaving 306720 bytes for local variables. Maximum is 327680 bytes.

gkowalewski, 2024-08-02:
[image]

gkowalewski, 2024-08-04:
It’s the same on Windows

Tinyu-Zhao, 2024-08-05:
This is FastLED library problem, please refer to this [issue](https://github.com/FastLED/FastLED/issues/1628)

zackees, 2024-08-05:
Here's the branch with a fix in it:

https://github.com/FastLED/FastLED#fix-esp32-s3-rmt-esp-idf-5

It's currently untested but the fix is super simple so better than vegas odds it works.

No idea how to use ArduinoIDE to use the FastLED branch with the fix. I use platformio because such issues become trivial to address with custom github sources for library dependencies. He's a test project you can use:

https://github.com/zackees/fastled_bug_s3_idf5x

zackees, 2024-08-09:
fix has been merged to 3.7.1. You can close this issue report.

closed 2024-08-09 by gkowalewski