m5stack/M5AtomS3 #28: Out of Date doesn't work with Arudino Current Version

RamboRogers, 2024-09-03:
### Describe the bug

Codes compiles but ESP32 throws this error when using demo code.

E (162) rmt(legacy): CONFLICT! driver_ng is not allowed to be usE (162) rmt(legacy): CONFLICT! driver_ng is not allowed to be used with the 

### To reproduce

Try to compile demo code on Arduino using current libraries.

### Expected behavior

Code works.

### Screenshots

[image]
[image]
[image]

### Environment

- OS: MacOS 
- IDE &IDE Version: Arduino 2.3.2
- Repository Version: M5AtomS3 0.1.2

### Additional context

It looks like M5 Unified compiles. Maybe it was fixed and this wasn't?

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Stack/issues?q=)
- [X] My report contains all necessary details

Tinyu-Zhao, 2024-09-04:
Are you using AtomS3 in the esp32 board or AtomS3 in the m5stack board?

RamboRogers, 2024-09-04:
You are right, I'm using the ESP32 board.  Thank you!

closed 2024-09-04 by RamboRogers