m5stack/M5AtomS3 #29: ATOMS3R Camera example fails with init error

truedat101, 2024-11-04:
### Describe the bug

I downloaded the code here.  I noticed the most recent commit in the repository for camera.ino adds a bug fix supposedly.

However, when run:

```
02:10:55.262 -> E (1142) cam_hal: cam_config(384): cam_dma_config failed
02:10:55.262 -> E (1143) camera: Camera config failed with error 0xffffffff
02:10:55.262 -> Camera Init Fail
02:10:56.253 -> ESP-ROM:esp32s3-20210327
02:10:56.253 -> Build:Mar 27 2021
02:10:56.253 -> rst:0xc (RTC_SW_CPU_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
```

This repeats endlessly.

### To reproduce

1. install camera.ino
2. view logs
3. camera gets init error repeatedly and crashes

### Expected behavior

should run, init cam, connect to wifi

### Environment

- OS:
- IDE &IDE Version:
- Repository Version:

### Additional context

Workaround: I think the instructions are missing, "Enable OPI PSRAM".  Put this into comments in camera.ino.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Stack/issues?q=)
- [X] My report contains all necessary details

Tinyu-Zhao, 2024-11-06:
Have you turned on PSRAM?

truedat101, 2024-11-06:
Yes, in my comment:

Workaround: I think the instructions are missing, "Enable OPI PSRAM". Put this into comments in camera.ino.

I maybe missed this if it was somewhere in the comments, but I didn't see it mentioned in the comments.

closed 2024-11-07 by Tinyu-Zhao