m5stack/M5AtomS3 #32: Can't read USB serial input

mgrouch, 2024-12-23:
### Describe the bug

Example of code with ESP32Console (https://github.com/jbtronics/ESP32Console)

````

#include <M5AtomS3.h>
#include <M5Unified.h>
#include <ESP32Console.h>
#include <FS.h>
#include <SPIFFS.h>
#include <ESP32Console/Helpers/PWDHelpers.h>

using namespace ESP32Console;

Console console;

void setup() {
  auto cfg = M5.config();
  AtomS3.begin(cfg);
  
  //Initalize SPIFFS and mount it on /spiffs
  //SPIFFS.begin(true, "/spiffs");

  //Modify prompt to show current file path (%pwd% get replaced by the filepath)
  //console.setPrompt("ESP32 %pwd%> ");
  console.setPrompt("m5> ");

  //Set HOME env for easier navigating (type cd to jump to home)
  setenv("HOME", "/spiffs", 1);
  //Set PWD to env
  //console_chdir("/spiffs");

  //Initialize Serial port on UART0 (the onboard USB Serial Converter)
  //Serial.begin(115200);

  //console.begin(115200, G2, G3, 0);
  console.begin(115200);
  //Serial.begin(115200);

  //Enable the saving of our command history to SPIFFS. You will be able to see it, when you type ls in your console.
  //console.enablePersistentHistory("/spiffs/.history.txt");

  console.registerSystemCommands();

  //Register the VFS specific commands
  //console.registerVFSCommands();

  //delay(2000);
  //Serial.flush();
}

void loop() {
  AtomS3.update();
  delay(100);
}

````

### To reproduce

````
#include <M5AtomS3.h>
#include <M5Unified.h>
#include <ESP32Console.h>
#include <FS.h>
#include <SPIFFS.h>
#include <ESP32Console/Helpers/PWDHelpers.h>

using namespace ESP32Console;

Console console;

void setup() {
  auto cfg = M5.config();
  AtomS3.begin(cfg);
  
  //Initalize SPIFFS and mount it on /spiffs
  //SPIFFS.begin(true, "/spiffs");

  //Modify prompt to show current file path (%pwd% get replaced by the filepath)
  //console.setPrompt("ESP32 %pwd%> ");
  console.setPrompt("m5> ");

  //Set HOME env for easier navigating (type cd to jump to home)
  setenv("HOME", "/spiffs", 1);
  //Set PWD to env
  //console_chdir("/spiffs");

  //Initialize Serial port on UART0 (the onboard USB Serial Converter)
  //Serial.begin(115200);

  //console.begin(115200, G2, G3, 0);
  console.begin(115200);
  //Serial.begin(115200);

  //Enable the saving of our command history to SPIFFS. You will be able to see it, when you type ls in your console.
  //console.enablePersistentHistory("/spiffs/.history.txt");

  console.registerSystemCommands();

  //Register the VFS specific commands
  //console.registerVFSCommands();

  //delay(2000);
  //Serial.flush();
}

void loop() {
  AtomS3.update();
  delay(100);
}
````

### Expected behavior

Should react to keyboard keys pressed in putty serial console connected to it at 115200 baud

### Environment

- OS:
- IDE &IDE Version:
- Repository Version:

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Stack/issues?q=)
- [X] My report contains all necessary details

mgrouch, 2024-12-23:
Related: https://github.com/jbtronics/ESP32Console/issues/19

https://github.com/jbtronics/ESP32Console/issues/7

Tinyu-Zhao, 2024-12-23:
Your problem probably has nothing to do with this warehouse. I suggest you go to ESP32Console for help.

Tinyu-Zhao, 2024-12-23:
Hope this helps you
https://github.com/jbtronics/ESP32Console/issues/8

closed 2024-12-23 by Tinyu-Zhao