m5stack/M5AtomS3 #34: driver/rmt_types.h: No such file or directory

majodi, 2025-01-16:
### Describe the bug

First use of the M5AtomS3R, following the guide here https://docs.m5stack.com/en/arduino/m5atoms3r/program

I get:

Alternatives for driver/rmt_types.h: []
ResolveLibrary(driver/rmt_types.h)
  -> candidates: []
In file included from /Users/majodi/dev/arduino2/libraries/FastLED/src/third_party/espressif/led_strip/src/led_strip.h:10,
                 from /Users/majodi/dev/arduino2/libraries/FastLED/src/platforms/esp/32/rmt_5/strip_rmt.cpp:12:
/Users/majodi/dev/arduino2/libraries/FastLED/src/third_party/espressif/led_strip/src/led_strip_rmt.h:12:10: fatal error: driver/rmt_types.h: No such file or directory
 #include "driver/rmt_types.h"
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: exit status 1

### To reproduce

In Arduino IDE Version: 2.3.4 with Board M5AtomS3R selected, display.ino (as guide instructs) - compile

### Expected behavior

No compile errors

### Environment

- OS: Macos Sonoma 14.6.1
- IDE &IDE Version: Arduino IDE Version: 2.3.4

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Stack/issues?q=)
- [x] My report contains all necessary details

Tinyu-Zhao, 2025-01-17:
Are you sure you chose AtomS3R in M5Stack instead of esp?
https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/docs/static/assets/img/arduino/atoms3r/quickstart_arduino_atoms3r_select_board.png

majodi, 2025-01-18:
Yes. But I can report that changing the board to ESP32S3 Dev Module does compile fine, see attached.

[image]
[image]

Tinyu-Zhao, 2025-01-20:
This should be a problem with the fastled rp.
They are not compatible with the arduino-esp32 v2.x version. You can try to delete this library and try again.

majodi, 2025-01-20:
After deleting Fastled library:

[image]

Tinyu-Zhao, 2025-01-20:
It is definitely a problem with the fastled warehouse. They did not consider the upgrade of the idf version. You can just use this FastLED 3.9.9 version.

You can ask this question in FastLED.

majodi, 2025-01-20:
Yes, FastLED 3.9.9 is working. But the latest version of FastLED is 3.9.11

You are using an older version of Arduino core which can not use newer libraries. So, you should change your documentation and tell people they only can use old libraries until you update your board file.

And, as a suggestion, also update the M5GFX documentation which is not up-to-date with the Lovyan implementation.

Tinyu-Zhao, 2025-01-21:
Our documentation engineer will add relevant content. Thank you, I will close the issue.

closed 2025-01-21 by Tinyu-Zhao